epan-utbm/utbm_robocar_dataset #1: No ground truth data

chenaaron0917, 2019-04-30:
I downloaded the bag file but there is no ros topic containing gps or ground truth data? Where can I find it?
Thanks

yzrobot, 2019-04-30:
Hi, 

The GPS-RTK (ground truth for localization) is ```/nmea_sentence```.

To use it:

* Download the driver [https://github.com/epan-utbm/utbm_robocar_dataset/tree/baselines/magellan_proflex500](https://github.com/epan-utbm/utbm_robocar_dataset/tree/baselines/magellan_proflex500) and put it in your ```catkin_ws/src/```

* Put this code block in your launch file:
```xml
  <node pkg="nmea_navsat_driver" type="nmea_topic_driver" name="nmea_topic_driver" output="screen"/>
  <node pkg="gps_common" type="utm_odometry_node" name="utm_odometry_node"/>
<node pkg="magellan_proflex500_driver" type="magellan_proflex500_driver" name="magellan_proflex500_driver"/>
```

Cheers,
Zhi

chenaaron0917, 2019-04-30:
Thank you.
Also, I tried to run the bag on 20180719 but the link provided is directed to 0713.

yzrobot, 2019-04-30:
Thx for the feedback. The link is fixed.

limeng1523, 2020-07-17:
Hi, @yzrobot  In baselines there is a package 'utbm_pose_estimation'. Can the output of this package be used as the groundtruth?

yzrobot, 2020-08-06:
Hi, I don't think so (as the name "estimation"),  the only one we can call “ground-truth” in the data we currently provide is the GPS-RTK, i.e. the `/fix` ROS topic. Cheers.

subshine, 2020-09-12:
> Hi,
> 
> The topic for GPS-RTK (ground truth for localization) is `/nmea_sentence`.
> 
> To use it:
> 
> * Download the driver https://github.com/epan-utbm/utbm_robocar_dataset/tree/baselines/drivers/magellan_proflex500 and put it in your `catkin_ws/src/`
> * Put this code block in your launch file:
> 
> ```
> <node pkg="nmea_navsat_driver" type="nmea_topic_driver" name="nmea_topic_driver"/>
> <node pkg="gps_common" type="utm_odometry_node" name="utm_odometry_node"/>
> <node pkg="magellan_proflex500_driver" type="magellan_proflex500_driver" name="magellan_proflex500_driver"/>
> ```
> 
> or check here: https://github.com/epan-utbm/utbm_robocar_dataset/tree/baselines/launch
> 
> Cheers,
> Zhi
Hi ,How to /nmea_sentence ros topic convert to /fix ,I want to get  gps data。
According to the above steps, it is not possible, can you be more detailed?
Thanks!

yzrobot, 2020-09-14:
Hi, hope this can help.
[image]

subshine, 2020-09-15:
> Hi, hope this can help.
> [image]

[image]
Thanx very much!Can you check the original dataset  echo / nmeasentence?  I echo /fix data is NULL,echo /nmeasentence is NULL,but echo /imu/data is normal.

subshine, 2020-09-15:
> Hi, hope this can help.
> [image]

solved.Thx! 
main problem lack of https://github.com/ros-drivers/nmea_msgs  package!

closed 2021-02-04 by yzrobot