epan-utbm/utbm_robocar_dataset #6: Cannot find imu/mag

MostafaOsman144, 2019-11-12:
When I check the bagfiles from the dataset, I cannot find the imu/mag data. Consequently, when I try to run the utbm_pose_estimation, /ubtm_pose_estimation/fix/pose is not being published.

yzrobot, 2019-11-13:
Could you clarify which bags do not contain `imu/mag`? I've just randomly checked two, it's there.

```
path:        utbm_robocar_dataset_20180719_noimage.bag
version:     2.0
duration:    15:26s (926s)
start:       Jul 19 2018 18:15:57.84 (1532016957.84)
end:         Jul 19 2018 18:31:24.51 (1532017884.51)
size:        4.8 GB
messages:    472470
compression: none [6106/6106 chunks]
types:       geometry_msgs/TwistStamped  [98d34b0043a2093cf9d9345ab6eef12e]
             nmea_msgs/Sentence          [9f221efc5f4b3bac7ce4af102b32308b]
             sensor_msgs/Imu             [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/LaserScan       [90c7ef2dc6895d81024acba2ac42f369]
             sensor_msgs/MagneticField   [2f3b0b43eed0c9501de0fa3ff89a45aa]
             sensor_msgs/PointCloud2     [1158d486dd51d683ce2f1be655c3c181]
             sick_ldmrs_msgs/ObjectArray [09128101facd48306fd0cf85eaf2be8f]
             std_msgs/String             [992ce8a1687cec8c8bd883ec73ca41d1]
             velodyne_msgs/VelodyneScan  [50804fc9533a0e579e6322c04ae70566]
topics:      /cloud                           11582 msgs    : sensor_msgs/PointCloud2    
             /hdl32e_left/velodyne_packets     9259 msgs    : velodyne_msgs/VelodyneScan 
             /hdl32e_right/velodyne_packets    9259 msgs    : velodyne_msgs/VelodyneScan 
             /imu/data                        92673 msgs    : sensor_msgs/Imu            
             /imu/mag                         92673 msgs    : sensor_msgs/MagneticField  
             /imu_data_str                    92673 msgs    : std_msgs/String            
             /nmea_sentence                   13901 msgs    : nmea_msgs/Sentence         
             /objects                         11579 msgs    : sick_ldmrs_msgs/ObjectArray
             /scan                            46197 msgs    : sensor_msgs/LaserScan      
             /velocity                        92674 msgs    : geometry_msgs/TwistStamped
```

```
path:        utbm_robocar_dataset_20190110_noimage.bag
version:     2.0
duration:    10:59s (659s)
start:       Jan 10 2019 10:06:15.10 (1547111175.10)
end:         Jan 10 2019 10:17:15.10 (1547111835.10)
size:        2.1 GB
messages:    320010
compression: none [2458/2458 chunks]
types:       geometry_msgs/TwistStamped  [98d34b0043a2093cf9d9345ab6eef12e]
             sensor_msgs/Imu             [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/LaserScan       [90c7ef2dc6895d81024acba2ac42f369]
             sensor_msgs/MagneticField   [2f3b0b43eed0c9501de0fa3ff89a45aa]
             sensor_msgs/PointCloud2     [1158d486dd51d683ce2f1be655c3c181]
             sick_ldmrs_msgs/ObjectArray [09128101facd48306fd0cf85eaf2be8f]
             std_msgs/String             [992ce8a1687cec8c8bd883ec73ca41d1]
             velodyne_msgs/VelodyneScan  [50804fc9533a0e579e6322c04ae70566]
topics:      /cloud                           8252 msgs    : sensor_msgs/PointCloud2    
             /hdl32e_left/velodyne_packets    6594 msgs    : velodyne_msgs/VelodyneScan 
             /imu/data                       66005 msgs    : sensor_msgs/Imu            
             /imu/mag                        66005 msgs    : sensor_msgs/MagneticField  
             /imu_data_str                   66005 msgs    : std_msgs/String            
             /objects                         8252 msgs    : sick_ldmrs_msgs/ObjectArray
             /scan                           32892 msgs    : sensor_msgs/LaserScan      
             /velocity                       66005 msgs    : geometry_msgs/TwistStamped
```

MostafaOsman144, 2019-11-13:
I checked utbm_robocar_dataset_20180502_noimage1.bag and utbm_robocar_dataset_20180502_noimage2.bag and both does not have imu/mag.

```
path:         utbm_robocar_dataset_20180502_noimage1.bag
version:      2.0
duration:     14:30s (870s)
start:        May 02 2018 14:40:27.68 (1525286427.68)
end:          May 02 2018 14:54:58.67 (1525287298.67)
size:         2.6 GB
messages:     207478
compression:  bz2 [5615/5615 chunks; 59.60%]
uncompressed: 4.4 GB @ 5.1 MB/s
compressed:   2.6 GB @ 3.1 MB/s (59.60%)
types:        nmea_msgs/Sentence             [9f221efc5f4b3bac7ce4af102b32308b]
              sensor_msgs/Imu                [6a62c6daae103f4ff57a132d6f95cec2]
              sensor_msgs/LaserScan          [90c7ef2dc6895d81024acba2ac42f369]
              sensor_msgs/NavSatFix          [2d3a8cd499b9b4a0249fb98fd05cfa48]
              sensor_msgs/PointCloud2        [1158d486dd51d683ce2f1be655c3c181]
              velodyne_msgs/VelodyneScan     [50804fc9533a0e579e6322c04ae70566]
              visualization_msgs/MarkerArray [d155b9ce5188fbaf89745847fd5882d7]
topics:       /cloud                           10890 msgs    : sensor_msgs/PointCloud2       
              /fix                              4355 msgs    : sensor_msgs/NavSatFix         
              /hdl32e_left/velodyne_packets     8581 msgs    : velodyne_msgs/VelodyneScan    
              /hdl32e_right/velodyne_packets    8581 msgs    : velodyne_msgs/VelodyneScan    
              /imu/data                        87105 msgs    : sensor_msgs/Imu               
              /nmea_sentence                    4355 msgs    : nmea_msgs/Sentence            
              /object_markers                  40208 msgs    : visualization_msgs/MarkerArray
              /scan                            43403 msgs    : sensor_msgs/LaserScan
```
and 

`path:         utbm_robocar_dataset_20180502_noimage2.bag
version:      2.0
duration:     13:55s (835s)
start:        May 02 2018 15:28:13.00 (1525289293.00)
end:          May 02 2018 15:42:08.08 (1525290128.08)
size:         2.5 GB
messages:     198795
compression:  bz2 [5383/5383 chunks; 59.68%]
uncompressed: 4.2 GB @ 5.1 MB/s
compressed:   2.5 GB @ 3.1 MB/s (59.68%)
types:        nmea_msgs/Sentence             [9f221efc5f4b3bac7ce4af102b32308b]
              sensor_msgs/Imu                [6a62c6daae103f4ff57a132d6f95cec2]
              sensor_msgs/LaserScan          [90c7ef2dc6895d81024acba2ac42f369]
              sensor_msgs/NavSatFix          [2d3a8cd499b9b4a0249fb98fd05cfa48]
              sensor_msgs/PointCloud2        [1158d486dd51d683ce2f1be655c3c181]
              velodyne_msgs/VelodyneScan     [50804fc9533a0e579e6322c04ae70566]
              visualization_msgs/MarkerArray [d155b9ce5188fbaf89745847fd5882d7]
topics:       /cloud                           10441 msgs    : sensor_msgs/PointCloud2       
              /fix                              4175 msgs    : sensor_msgs/NavSatFix         
              /hdl32e_left/velodyne_packets     8227 msgs    : velodyne_msgs/VelodyneScan    
              /hdl32e_right/velodyne_packets    8228 msgs    : velodyne_msgs/VelodyneScan    
              /imu/data                        83513 msgs    : sensor_msgs/Imu               
              /nmea_sentence                    4175 msgs    : nmea_msgs/Sentence            
              /object_markers                  38393 msgs    : visualization_msgs/MarkerArray
              /scan                            41643 msgs    : sensor_msgs/LaserScan`

yzrobot, 2019-11-14:
Sorry, you are right, `imu/mag` is missing in these two bags....

closed 2021-02-04 by yzrobot