epan-utbm/utbm_robocar_dataset #7: CSV Files for Ground Truth

MostafaOsman144, 2019-11-13:
I can't find the Groundtruth CSV files of any bagfiles except for GPS-RTK-20180719. In another issue, it is mentioned that the ground truth is in nemea_sentence, how can I convert the string inside these messages to xyz coordinates similar to those in the CSV files ?

yzrobot, 2019-11-13:
There's only one CSV file, you need to generate the others by yourself. With:
```
  <!-- Magellan ProFlex 500 GNSS Receiver -->
  <node pkg="nmea_navsat_driver" type="nmea_topic_driver" name="nmea_topic_driver" output="screen"/>
  <node pkg="gps_common" type="utm_odometry_node" name="utm_odometry_node"/>
  <node pkg="magellan_proflex500_driver" type="magellan_proflex500_driver" name="magellan_proflex500_driver"/>
```
you can get GT position from `/magellan_proflex500_driver/marker_array` and easy to save as CSV file.

MostafaOsman144, 2019-11-13:
Ok, I find it now, thank you.

closed 2019-11-13 by MostafaOsman144

subshine, 2020-09-12:
> There's only one CSV file, you need to generate the others by yourself. With:
> 
> ```
>   
>   <node pkg="nmea_navsat_driver" type="nmea_topic_driver" name="nmea_topic_driver" output="screen"/>
>   <node pkg="gps_common" type="utm_odometry_node" name="utm_odometry_node"/>
>   <node pkg="magellan_proflex500_driver" type="magellan_proflex500_driver" name="magellan_proflex500_driver"/>
> ```
> 
> you can get GT position from `/magellan_proflex500_driver/marker_array` and easy to save as CSV file.

When I launch the launch file and play bag, but I cannot echo /fix  or /magellan_proflex500_driver/marker_array data through the ros topic,how to deal with?

yzrobot, 2020-09-14:
Hi, if you are using this https://github.com/epan-utbm/utbm_robocar_dataset/tree/baselines/launch, you should be able to see:
```
$ rostopic echo /fix 
header: 
  seq: 540
  stamp: 
    secs: 1532017068
    nsecs: 403332926
  frame_id: "navsat"
status: 
  status: 2
  service: 1
latitude: 47.51322485
longitude: 6.79003481667
altitude: 412.084
position_covariance: [0.81, 0.0, 0.0, 0.0, 0.81, 0.0, 0.0, 0.0, 3.24]
position_covariance_type: 1
```
```
$ rostopic echo /magellan_proflex500_driver/marker_array
markers: 
  - 
    header: 
      seq: 11
      stamp: 
        secs: 1532016960
        nsecs: 401259270
      frame_id: "navsat"
    ns: ''
    id: 0
    type: 4
    action: 0
    pose: 
      position: 
        x: 0.0
        y: 0.0
        z: 0.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    scale: 
      x: 0.1
      y: 0.0
      z: 0.0
    color: 
      r: 1.0
      g: 0.5
      b: 0.0
      a: 1.0
    lifetime: 
      secs: 1
      nsecs:         0
    frame_locked: False
    points: 
      - 
        x: 0.0
        y: 0.0
        z: 0.0
      - 
        x: 0.0
        y: 0.0
        z: 0.0
      - 
        x: 0.0
        y: 0.0
        z: 0.000999999999976
      - 
        x: 0.00114960712381
        y: -0.00373996421695
        z: -0.00200000000001
      - 
        x: -0.00109697657172
        y: 0.00559211988002
        z: -0.00100000000003
      - 
        x: 0.00130749872187
        y: 0.00181649718434
        z: 0.000999999999976
      - 
        x: -0.0011496070656
        y: 0.00373996514827
        z: -0.00100000000003
      - 
        x: 0.000105261104181
        y: 0.00370430760086
        z: -0.00100000000003
      - 
        x: 0.000157891598064
        y: 0.00555646140128
        z: 0.0
      - 
        x: -0.00365934561705
        y: 0.00381128210574
        z: 0.00200000000001
      - 
        x: -0.00491421384504
        y: 0.00384693965316
        z: 0.00200000000001
      - 
        x: -0.00622171256691
        y: 0.00203044526279
        z: 0.000999999999976
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
---
```

subshine, 2020-09-15:
> Hi, if you are using this https://github.com/epan-utbm/utbm_robocar_dataset/tree/baselines/launch, you should be able to see:
>
> ```
> $ rostopic echo /fix
> header:
>   seq: 540
>   stamp:
>     secs: [...]

Thank you very much!

YangSiri, 2021-06-29:
HI, @yzrobot 

When i launch the following

> There's only one CSV file, you need to generate the others by yourself. With:
>
> ```
>
>   <node pkg="nmea_navsat_driver" type="nmea_topic_driver" name="nmea_topic_driver" output="screen"/>
>   <node [...]

There is an error called 
"ERROR: cannot launch node of type [magellan_proflex500_driver/magellan_proflex500_driver]: magellan_proflex500_driver
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/cyz/workspace/catkin_main/src
ROS path [2]=/opt/ros/melodic/share"

so the node is not actived. What may be the problem ?

Looking forward to your help;)

yzrobot, 2021-06-29:
Hi, do you have the corresponding driver installed? Please check [here](https://github.com/epan-utbm/utbm_robocar_dataset/tree/baselines/drivers/magellan_proflex500) and [here](https://github.com/epan-utbm/utbm_robocar_dataset/tree/baselines/launch).

YangSiri, 2021-06-29:
Yes, @yzrobot 

The __magellan_proflex500_driver__ is built as well as the others.