epan-utbm/utbm_robocar_dataset #8: Calibration b/w IMU and GPS

SubMishMar, 2020-04-02:
Is it correct to assume that the calibration between IMU and GPS is identity? 

Can you also provide the intrinsic calibration parameters of the IMU used?

SubMishMar, 2020-04-02:
I think it is identity, as I can guess it from here:
https://github.com/epan-utbm/utbm_robocar_dataset/blob/6d50b7999260f1c5a3dd13360954468edc7de4b0/launch/utbm_dataset_play.launch#L85-L93

But I still want to know the IMU intrinsics, is it is available with you.

yzrobot, 2020-04-02:
Hi, the GPS antenna is mounted on the roof and the IMU is in the car. Both are placed on the z-axis perpendicular to the car rear axle. We could provide the intrinsic calibration parameters, but the university is close due to COVID-19. we have noted this and will update you ASAP.

yzrobot, 2020-06-25:
Hi, could you please check if (this)[https://github.com/epan-utbm/utbm_robocar_dataset/blob/baselines/imu/imu-test.bag] is what you need?

closed 2021-02-04 by yzrobot