epan-utbm/utbm_robocar_dataset #15: Can not find gps velocity?

libing64, 2020-10-25:


yzrobot, 2020-10-25:
Hi, did you try to run [this](https://github.com/epan-utbm/utbm_robocar_dataset/tree/baselines/launch)? If so, when you type `$ rostopic echo /vel`, you could get something like:
```
---
header: 
  seq: 6
  stamp: 
    secs: 1531487932
    nsecs: 969429833
  frame_id: "navsat"
twist: 
  linear: 
    x: 7.3565107369
    y: -0.0256791710611
    z: 0.0
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0
---
header: 
  seq: 7
  stamp: 
    secs: 1531487933
    nsecs: 160320017
  frame_id: "navsat"
twist: 
  linear: 
    x: 7.35403464769
    y: 0.192572174502
    z: 0.0
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0
```

Also c.f. [this](https://github.com/epan-utbm/utbm_robocar_dataset/blob/baselines/images/hector_rosgraph.png)

libing64, 2020-10-26:
@yzrobot Great thanks, so i need the nmea_topic_driver to calclulate gps velocity from nmea_sentence, right?
I would have a try.

closed 2020-10-26 by libing64