epan-utbm/utbm_robocar_dataset #17: Are those cameras synchronized?

hitdshu, 2020-12-04:
Thanks for your data. 

Are those cameras synchronized? Looking forward to your reply.

yzrobot, 2020-12-04:
Hi, the data are timestamped by ROS and the synchronization can be done at the software level (c.f. (http://wiki.ros.org/message_filters#Time_Synchronizer)[http://wiki.ros.org/message_filters#Time_Synchronizer]). However, it is not synchronized at the hardware level (i.e. with triggers when recording data).

hitdshu, 2020-12-04:
> Hi, the data are timestamped by ROS and the synchronization can be done at the software level (c.f. http://wiki.ros.org/message_filters#Time_Synchronizer). However, it is not synchronized at the hardware level (i.e. with triggers when recording data).

Thanks for your kind reply. I've read your paper and knew that it's not hardware synchronized. I actually want to know: Can the images be synchronized in a reasonable level and so can be seen as hardware synchronized? Thanks very much.

closed 2020-12-07 by hitdshu