epan-utbm/utbm_robocar_dataset #18: file not found

yuxu1997, 2020-12-18:
excuse me ,when i open the dataset download link (https://epan-utbm.github.io/utbm_robocar_dataset/), it shows that "file not found,The document could not be found on the server. Maybe the share was deleted or has expired?"

yzrobot, 2020-12-19:
Hi, sorry for the inconvenience, I just checked it urgently, and it seems to be something wrong with the cloud data storage server hosted by our uni. However, we are now in the Christmas holiday, I will deal with it ASAP after the holiday with our technicians.

yuxu1997, 2020-12-20:
> Hi, sorry for the inconvenience, I just checked it urgently, and it seems to be something wrong with the cloud data storage server hosted by our uni. However, we are now in the Christmas holiday, I will deal with it ASAP after the holiday with our technicians.

OK, thank you very much

closed 2020-12-20 by yuxu1997