epan-utbm/utbm_robocar_dataset #19: the datasets are not complete and timestamps don't match

yuxu1997, 2021-01-06:
I'm sorry to bother, but i find the image data are not complete at 2018-05-02. And the timstamps of image data and other sensor data at 2018-05-02 can not match. Besides, the server seems broken again, because i got the "404 Not Found". I'm hoping for your help, thanks.

yzrobot, 2021-01-11:
Hi, the server was in maintenance mode, now it's OK. We found that some bags were missing, and now we have recovered some of them (2018-05-02 evening is complete now) and are uploading them (so please check the links later), and we are trying to find the others and will keep you updated. The timestamps of the topics have been adjusted using [add_header_time_offset.zip](https://github.com/epan-utbm/utbm_robocar_dataset/files/5794581/add_header_time_offset.zip) (so you could further change it if needed). Cheers

yuxu1997, 2021-01-12:
> Hi, the server was in maintenance mode, now it's OK. We found that some bags were missing, and now we have recovered some of them (2018-05-02 evening is complete now) and are uploading them (so [...]

OK,Thanks

closed 2021-01-12 by yuxu1997