epan-utbm/utbm_robocar_dataset #20: Why the topic '/nmea_sentence' is not processed by the node '/nmea_topic_driver'?

robotstar, 2021-01-22:
I perform following operations:
$ roslaunch utbm_dataset_play.launch bag:=utbm_robocar_dataset_20180719_noimage.bag
$ rostopic echo /nmea_sentence  
WARNING: no messages received and simulated time is active.
Is /clock being published?
$ rqt_graph
[image]

It seems like that the topic '/nmea_sentence' is not processed by the node '/nmea_topic_driver'. Why it happens?

yzrobot, 2021-01-22:
Hi, to find out the problem, could you pls confirm with me first:
* roslaunch utbm_dataset_play.launch bag:=/FULL_PATH_TO_YOUR/utbm_robocar_dataset_20180719_noimage.bag
* you did  `sudo apt install ros-VERSION-nmea-navsat-driver`

robotstar, 2021-01-22:
> Hi, to find out the problem, could you pls confirm first:
> 
> * roslaunch utbm_dataset_play.launch bag:=/FULL_PATH_TO_YOUR/utbm_robocar_dataset_20180719_noimage.bag
> * you did  `sudo apt install ros-VERSION-nmea-navsat-driver`

Yes, I performed following operations following https://github.com/epan-utbm/utbm_robocar_dataset/tree/baselines/launch.
$ sudo apt-get install ros-melodic-velodyne
$ sudo apt-get install ros-melodic-nmea-navsat-driver
$ sudo apt-get install ros-melodic-gps-common
Besides, my Ubuntu version is 18.04.

robotstar, 2021-01-22:
> Hi, to find out the problem, could you pls confirm first:
> 
> * roslaunch utbm_dataset_play.launch bag:=/FULL_PATH_TO_YOUR/utbm_robocar_dataset_20180719_noimage.bag
> * you did  `sudo apt install ros-VERSION-nmea-navsat-driver`

And I used absolute path of the bag. The velodyne points can be visualized in the rviz. However, the node '/nmea_topic_driver' did not process the topic '/nmea_sentence'.

robotstar, 2021-01-22:
> Your `rosbag_play` output looks different from mine ....
> [image]

Yes. I am confused by my `rqt_graph` result. Why the topic '/nmea_sentence' disappeared in my `rosbag_play`?

robotstar, 2021-01-22:
> > Your `rosbag_play` output looks different from mine ....
> > [image]
> 
> Yes. I am confused by my `rqt_graph` result. Why the topic '/nmea_sentence' disappeared in my `rosbag_play`?

And the operation
$ rostopic echo /nmea_sentence
can not output the message in the terminal. Just output:
WARNING: no messages received and simulated time is active.
Is /clock being published?

lipascal123, 2021-03-06:
I got the same issue, any updates on this?