epan-utbm/utbm_robocar_dataset #21: Can i use raw GPS data only?

Gaopengen, 2021-02-03:
I noticed most GPS data is corrected by RTK, how can i get gps data only?
I also noticed that the data taken on 2019-01-31 does not contain RTK, while how to get the ground-truth of this sequence?  thanks!

yzrobot, 2021-02-04:
Hi, as you can see, most GPS data has been corrected by RTK, I'm sorry but basically we have no way to obtain. Regarding the data on the day of 2019-01-10, due to the snow that day, we were unable to drive the car and thus take the RTK base station to the top of the hill, so only GPS data was available. If you want to get the "precise" trajectory of the car, one possible way would be to use lidar odometry like [LeGO-LOAM](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM) or [LIO-SAM](https://github.com/TixiaoShan/LIO-SAM) or similar. Cheers.

Gaopengen, 2021-02-05:
> Hi, as you can see, most GPS data has been corrected by RTK, I'm sorry but basically we have no way to obtain the raw GPS data. Regarding the data on the day of 2019-01-31, due to the snow that day, we were unable to drive the car and thus take the RTK base station to the top of the hill, so only GPS data was available. If you want to get the "precise" trajectory of the car, one possible way would be to use lidar odometry like [LeGO-LOAM](https://github.com/RobustFieldAutonomyLab/LeGO-LOAM) or [LIO-SAM](https://github.com/TixiaoShan/LIO-SAM) or similar. Cheers.

Thanks for your reply!