epan-utbm/utbm_robocar_dataset #23: A question about ground truth

duanxz0127, 2021-11-27:
Hi, I output my SLAM-estimated attitude and the position in "magellan_proflex500_driver" separately, and the two trajectories seem to differ by a rotation around the z-axis. How can I solve this?
By the way, how can I quantify the difference between the two trajectories and thus evaluate the accuracy of my trajectory?

duanxz0127, 2021-11-27:
[image]

ShuyueLin, 2023-04-03:
Hi, may I ask you a question？When I subscribed to the ground truth from the topic /magellan_proflex500_driver/marker_array, I found that both x, y, z, and orientation were zero. Why is this?

jorgeffonte, 2024-08-03:
> Hi, I output my SLAM-estimated attitude and the position in "magellan_proflex500_driver" separately, and the two trajectories seem to differ by a rotation around the z-axis. How can I solve this? By [...]

Did you mange to get correct rotation?

jorgeffonte, 2024-08-06:
> Hi, may I ask you a question？When I subscribed to the ground truth from the topic /magellan_proflex500_driver/marker_array, I found that both x, y, z, and orientation were zero. Why is this?

The nmea_sentence message does not provide heading information, so orientations are not obtained. Therefore, rotation evaluation metrics cannot be obtained either.