epan-utbm/utbm_robocar_dataset #25: Failed to load nodelet [/hdl32e_left/cloud_nodelet] of type [velodyne_pointcloud/CloudNodelet]

yuhui-xie, 2022-03-29:
I have successfully run it about 2 months ago, but today I want try it and find errors:
[ERROR] [1648522310.752299538]: Failed to load nodelet [/hdl32e_left/cloud_nodelet] of type [velodyne_pointcloud/CloudNodelet] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type velodyne_pointcloud::CloudNodelet as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1648522310.752338509]: The error before refreshing the cache was: MultiLibraryClassLoader: Could not create object of class type velodyne_pointcloud::CloudNodelet as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[FATAL] [1648522310.752467228]: Failed to load nodelet '/hdl32e_left/cloud_nodelet` of type `velodyne_pointcloud/CloudNodelet` to manager `velodyne_left_nodelet_manager'
[ERROR] [1648522310.768457307]: Failed to load nodelet [/hdl32e_right/cloud_nodelet] of type [velodyne_pointcloud/CloudNodelet] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type velodyne_pointcloud::CloudNodelet as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1648522310.768493485]: The error before refreshing the cache was: MultiLibraryClassLoader: Could not create object of class type velodyne_pointcloud::CloudNodelet as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[FATAL] [1648522310.768778300]: Failed to load nodelet '/hdl32e_right/cloud_nodelet` of type `velodyne_pointcloud/CloudNodelet` to manager `velodyne_right_nodelet_manager'
[hdl32e_left/cloud_nodelet-4] process has died [pid 20513, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load velodyne_pointcloud/CloudNodelet velodyne_left_nodelet_manager __name:=cloud_nodelet __log:=/home/yuhui/.ros/log/2e3dc058-af0b-11ec-9e56-e0d4e8c8c26f/hdl32e_left-cloud_nodelet-4.log].
log file: /home/yuhui/.ros/log/2e3dc058-af0b-11ec-9e56-e0d4e8c8c26f/hdl32e_left-cloud_nodelet-4*.log
[hdl32e_right/cloud_nodelet-6] process has died [pid 20520, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load velodyne_pointcloud/CloudNodelet velodyne_right_nodelet_manager __name:=cloud_nodelet __log:=/home/yuhui/.ros/log/2e3dc058-af0b-11ec-9e56-e0d4e8c8c26f/hdl32e_right-cloud_nodelet-6.log].
log file: /home/yuhui/.ros/log/2e3dc058-af0b-11ec-9e56-e0d4e8c8c26f/hdl32e_right-cloud_nodelet-6*.log

it seems that velodyne relevant pkgs meet some problems?
But I really have install ros-melodic-velodyne as readme provided.

william-zhangshuai, 2022-04-17:
Hello, have you solved this problem?

jorgeffonte, 2024-07-31:
try to change on [play launch](https://github.com/epan-utbm/utbm_robocar_dataset/blob/3e229a80c2fe13ef82f7e63bb65324e5de4d1e7c/launch/utbm_dataset_play.launch#L11)
from
`args="load velodyne_pointcloud/CloudNodelet velodyne_right_nodelet_manager"`
to
`args="load velodyne_pointcloud/TransformNodelet velodyne_right_nodelet_manager"`